mesh-iit/documentation #16: Add missing info on ele and wiring kits to Upperarm Covers UKIT

pattacini, 2020-07-06:
The Upperam Covers UKIT documentation lacks the required information about the missing sub-kits (`ele` and `wiring`), which have been already added into our management system.

cc @Elio75

Elio75, 2020-07-06:
In the documentation made # 495, I think we could add the respective codes to WKIT and EKIT in the Ukit table and what they contain, it seems to me that these also form part of the whole assembly of the upgrade upper arm cover?

[image]

pattacini, 2020-07-06:
Yes indeed @Elio75, would you be willing to amend the documentation then?
Just do what we've done together via fork and pull-request.

Elio75, 2020-07-06:
Ok i will do it as soon as possible

Elio75, 2020-07-14:
I added in the UKIT table of documentation the EKIT_004 and WKIT_004 with its respective contents.
I await the return of Emiliano Barbieri to make me give some pictures of the covers with all the electronics mounted.

[image]
[image]

pattacini, 2020-07-14:
Cool thanks @Elio75 👍 
We'll be awaiting the final PR then.

pattacini, 2020-07-24:
PR merged: see #27 🚀

closed 2020-07-24 by pattacini