mesh-iit/documentation #17: Add missing code to identify the Upperarm covers UKIT (Upperarms, fix skin breaking)

Fabrizio69, 2020-07-06:
## Task Description
The code to identify the global assembly to upgrade the upperarm is missing

## Definition of Done
- [ ] The documentation contains the upgrade identification code
- [ ] Each part that constitutes the UKIT is correctly identified
- [ ] The scope of the upgrade is clearly specified

pattacini, 2020-07-24:
@Fabrizio69 now the documentation of UKIT should be more nice-looking from your standpoint.
Give it a glance and report here any feedback you may have, please.

traversaro, 2020-07-24:
I am not dealing with UKIT on a daily basis, so I may suggest something that it may not have sense, but if the UKIT code is the main entry point and identified for a ukit, I think it could make sense to add it to the summary and the menu to access the UKITs: 
[image]

Because I imagine that in the future the number of UKITs will grow, and opening all the pages of the UKITs to understand to which human readable name corresponds the UKIT 003 may be complex.

pattacini, 2020-07-24:
I'll do that 👍🏻

pattacini, 2020-07-24:
Done @traversaro 👍🏻

pattacini, 2020-07-27:
The information you required is now available @Fabrizio69.
Closing for now, you could still point us at any improvements you may have.

closed 2020-07-27 by pattacini