mesh-iit/documentation #71: Doubts on icub_operating_systems section

traversaro, 2021-01-11:
Today with @Iaxama we used the icub_operating_systems  and we had a few doubts. I would be happy to go on and open a PR fixing this aspects, but I am afraid I am not sure about some information on this, as mentioned in the following.

### "real" icub server

The iCub server section mention to options, one is the "real" iCub server, while another is the "laptop used as the iCub server" .
From the context, it is not clear what is the meaning of the "real" iCub server. I guess it is meant for when the iCub server is installed not on a laptop?

[image]

### iCubOS versions
I was not able to understand on which version of Ubuntu the iCubOS is based. From https://icub-tech-iit.github.io/documentation/icub_operating_systems/ it seems that `iCubOS` is used to refer to the OS image meant to be running on the `icub-head` machine that is the COMExpress board that is placed on the iCub head, and from https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/icubos/ it seems that it is based on Ubuntu 18.04 . 
But then in https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-from-image/ it seems that the iCubOS image is also used for the iCub server, and from https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-from-scratch/ it seems that this image should be based on Ubuntu 20.04. 
I guess I am missing something, but I am not sure what.

traversaro, 2021-01-11:
fyi @mbrunettini @davidelasagna

traversaro, 2021-01-12:
fyi @Iaxama @2103simon

pattacini, 2021-01-18:
Any news on this @mbrunettini ?

mbrunettini, 2021-01-21:
- The "real" iCub Server is the one used in all our setups, with a dedicated machine acting only as server setup
- The laptop server is the one provided in those setups with only a single machine acting  both as server (gateway, ntp and NFS) and console (this is the "basic" iCub setup we sell)

mbrunettini, 2021-01-21:
Regarding OS running on the robot:
- the current version for the PC104 (diskless) is iCubLive-OS 7.10 based on Debian 10 live, [see](https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/icub-live/)
- the current version for icub-head (with disk) is iCub-OS 8.18.04 based on Ubuntu server 18.04 [see](https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/icubos/)

traversaro, 2021-01-21:
Thanks @mbrunettini ! Do you think we can add this information on the two different kind of server setup in the docs, for example in https://icub-tech-iit.github.io/documentation/icub_operating_systems/ or where do you think it may be more appropriate?

mbrunettini, 2021-01-21:
@traversaro 
Regarding this

> Regarding OS running on the robot:
> 
> the current version for the PC104 (diskless) is iCubLive-OS 7.10 based on Debian 10 live, see this
> the current version for icub-head (a.k.a. COMExpress, with disk) is iCub-OS 8.18.04 based on Ubuntu server 18.04 see this

I think the information is already [there ](https://icub-tech-iit.github.io/documentation/icub_operating_systems/)

mbrunettini, 2021-01-21:
@traversaro 
While regarding the iCub Server, we can improve this section of the manual (same page as above, but in the bottom part):

> Once you customized your machine, you can proceed with further setup for special machines, such as:
> 
> CUDA workstation
> The iCub Server (Workstation)
> The iCub Server (laptop)

mbrunettini, 2021-01-21:
I changed the [above page](https://icub-tech-iit.github.io/documentation/icub_operating_systems/)

pattacini, 2021-02-02:
Can we close this?

pattacini, 2021-02-03:
Closing, feel free to open it up again if needed.

closed 2021-02-03 by pattacini

traversaro, 2021-02-03:
Yes, sorry but apparently I had lost a bunch of notifications from this repo.