mesh-iit/documentation #73: Update icub_operating_systems/other-machines/generic-machine.md for Ubuntu 20.04 support

traversaro, 2021-01-12:
The documentation in https://github.com/icub-tech-iit/documentation/blob/master/docs/icub_operating_systems/other-machines/generic-machine.md#static-ip-configuration refers to the `/etc/network/interfaces` file, that from what I understand (@Iaxama @2103simon) it is not supported anymore on Ubuntu 20.04 . Probably we should modify that section to be Ubuntu 20.04 . 

Related issue: https://github.com/robotology/robotology-superbuild/issues/481 .

Iaxama, 2021-01-14:
With Ubuntu 20.04 i had to set up the network using the nm-connection-editor (in particular to specify the search domains). This might not work for headless machines though.

pattacini, 2021-01-18:
Comments @mbrunettini ?

mbrunettini, 2021-01-21:
We have two options: 
- use NTEPLAN as in headless machines (see [this part of the manual](https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/networking/) 
- Use NetworkMonitor GUI (the common Ubuntu desktop tool)

I'll update the guide with the above two options.

@laxama : Personally, I've never used nm-connection-editor

pattacini, 2021-02-02:
Any update on this?

mbrunettini, 2021-02-02:
The migration to Ubuntu 20.04 is in progress, see icub-tech-iit/tickets#685

closed 2021-02-08 by mbrunettini