mesh-iit/documentation #80: Improve documentation on network configurations to connect laptop to EMS board

isorrentino, 2021-01-25:
I needed to connect my laptop to the motor setup in DIC area and I discovered that we do not have public and official documentation on how to do it. In particular, I discovered that I had to set a static IP address and a certain subnet mask to the network interface to be able to communicate with the EMS board. This information is not anywhere and future users could experience problems in setting up the correct configuration.

traversaro, 2021-01-25:
Some useful network docs are provided in https://icub-tech-iit.github.io/documentation/icub_firmware/firmware/firmware/#the-topology-of-the-robot , but indeed the subnet mask is not listed there nor in any of the `network.<YARP_ROBOT_NAME>.xml` files referenced. Probably that could be a good point in which to add a reference to the subnet mask used, what do you think @davidetome ?

davidetome, 2021-01-25:
@traversaro I take this in charge 📝

cc @pattacini

traversaro, 2021-01-25:
Thanks @davidetome !

davidetome, 2021-02-05:
Done, please see this [PR](https://github.com/icub-tech-iit/documentation/pull/83)

closed 2021-02-05 by pattacini