mesh-iit/documentation #85: change title for "configure-static-ip"

mbrunettini, 2021-02-11:
The title for the page https://icub-tech-iit.github.io/documentation/configure_static_ip/configure-static-ip/
can be very misleading, please specify better that this case applies ONLY to EMS communication

pattacini, 2021-02-11:
@davidetome could you please have a look at this?

davidetome, 2021-02-11:
> The title for the page https://icub-tech-iit.github.io/documentation/configure_static_ip/configure-static-ip/
> can be very misleading, please specify better that this case applies ONLY to EMS communication

@mbrunettini It is clearly specified in the doc description content at top of the page  that the doc applies to all ETH boards (not only EMS); anyway I'm going to change the title too 👍🏻 

cc @pattacini

mbrunettini, 2021-02-11:
Sure, BUT I was referring to the title shown on the left navigation bar, at that level is very _visible_ and  can be misleading

davidetome, 2021-02-11:
done in this [PR](https://github.com/icub-tech-iit/documentation/pull/87)

closed 2021-02-11 by pattacini