mesh-iit/documentation #89: Broken links in iCub wiki

alecive, 2021-02-22:
I hope this 2021 is off to a good start for the iCub Techs. I just wanted to raise the issue that there may be broken links in the iCub wiki. For example:

-  the (F/T sensors page)[http://wiki.icub.org/wiki/FT_sensor] points the user to https://icub-tech-iit.github.io/documentation/icub_firmware/ft-sensors/ft-sensors/ which is a badly formatted page with no clear way to get to the correct link (which should be (this one)[https://icub-tech-iit.github.io/documentation/ft-sensors/ft-sensors/])
- One other thing I recommend doing is to have this link https://icub-tech-iit.github.io point to the homepage of your documentation (i.e. this one https://icub-tech-iit.github.io/documentation/ ). As of now, the link above displays just a generic jekyll template and it is very hard to get to the right link unless you know it.
- Finally, some good SEO would be nice. Right now, it's basically impossible to find this new documentation website if you are not aware of it. There is a handy `jekyll-seo` package that is supported by github and should make the job easy!

I hope this helps. I am tagging @julijenv just because 😎

julijenv, 2021-02-22:
Thank you so much for your feedback @alecive ,
I am sure someone will take good care of this and may be trying your suggestion about 'jekyll-seo', unfirtunately I am not responsible  for doing this myself, I am just a little contributor of this new wiki, which we hope to give more handful tool and details to repair iCub.
thanks one more time Ale

traversaro, 2021-02-22:
> Finally, some good SEO would be nice. Right now, it's basically impossible to find this new documentation website if you are not aware of it. There is a handy `jekyll-seo` package that is supported by github and should make the job easy!

FYI website is using [mkdocs](https://www.mkdocs.org/) and not Jekyll, but I guess probably something similar  also exists for mkdocs, a quick google search found https://squidfunk.github.io/mkdocs-material/reference/meta-tags/  .

pattacini, 2021-02-22:
Hey @alecive 😀
Hope you're doing well dude! Thanks for reporting!

1. Fixed ✔ - we moved that page recently w/o updating the pointers in the "old" wiki.
2. Nope ❌- https://icub-tech-iit.github.io/ has nothing to do w/ https://icub-tech-iit.github.io/documentation/
3. You've got to know our new [Bazaar](https://robot-bazaar.iit.it/) as well as https://icub.iit.it! There are links there pointing to the documentation.

alecive, 2021-02-22:
@julijenv I just tagged you to nag you :-) I hope you are doing well! 

The new bazaar and icub website are great! I still find it hard to navigate all the websites and understand how they are connected (i.e. icub wiki vs icub.org vs robotcub.org vs icub.iit.it vs bazaar vs icub-tech-iit.github.io vs probably something else I'm missing), but I enjoy the modernization :-) 

Why don't you just have  https://icub-tech-iit.github.io/ redirect to w/ https://icub-tech-iit.github.io/documentation/ ? It will simplify navigation and discovery. Anyways, fine by me!

pattacini, 2021-02-22:
Just deleted https://icub-tech-iit.github.io: it was useless, actually. We will restore it when needed.

We are slowly phasing out https://wiki.icub.org/wiki/Main_Page in favor of more modern resources; this takes quite some time as you can easily figure out.

I didn't even remember about http://robotcub.org 😄

closed 2021-02-23 by pattacini