mesh-iit/documentation #98: Possible missing information about PC104 live image (when combined with robolotogy-superbuild)

xEnVrE, 2021-04-16:
I am opening this issue to discuss _possible_ missing information about the iCub PC104 live image in a scenario where we use the `robotology-superbuild` (but not necessarily limited to this actually).

### 1. CMake version

The latest `robotology-superbuild` [requires](https://github.com/robotology/robotology-superbuild#system-dependencies) CMake 3.16. If I am not wrong the latest PC104 image is based on Debian Buster (10) that [has](https://repology.org/project/cmake/versions), by default, CMake 3.13.4. According to the `robotology-superbuild` documentation it is possible to update on the latest by enabling `buster-backports`.

My question is, then, should this step be done by the user after installing the PC104 image on the robot? Is it the case that we add something about in the documentation such that the scenario in which the user uses the superbuild is covered? Another solution, but I don't know if is feasible, would be to pre-configure the PC104 image such that backports are enabled.

### 2. diagnostic daemon

The superbuild, but I think this is independent from the superbuild itself, enables build of the [diagnostic daemon](https://github.com/robotology/diagnostic-daemon) library when the profile `ROBOTOLOGY_ENABLE_ICUB_HEAD` is enabled. As stated in the repository of the library, it requires `libboost-system-dev`. However, as per [the documentation](https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/the-icub-live-details/#additional-packages), the PC104 live image does not include this package by default.

My question is similar in this case. Should we add something in the documentation to cover this or is it the case to ship the live image pre-configured with that package?

Thank you

cc @pattacini @traversaro @mbrunettini

pattacini, 2021-04-16:
My hunch is that problem number 2 should be tackled upstream by installing the missing library in the live image.

Because of the need for updating the live image, then we can also address problem number 1 by installing CMake 3.16.
This way, we won't need either to change the documentation or to force the user to do that on his own.

That said, we might somehow enforce the alignment between the superbuild requirements and the procedure that allows yielding the live image?

What are your thoughts?

traversaro, 2021-04-16:
> That said, we might somehow enforce the alignment between the superbuild requirements and the procedure that allows yielding the live image?

Yes, on this specific case I forgot to report upstream this requirement so that they could be integrated in the image.

mbrunettini, 2021-04-16:
I'll try to update the image and let you know

pattacini, 2021-04-16:
Thanks heaps @mbrunettini 👍🏻

mbrunettini, 2021-04-22:
I produced a new PC104 OS image with the following upgrade:
- latest packages release
- `cmake `3.16.3
- `libboost-system-dev`

See https://github.com/icub-tech-iit/icub-os-files/tree/master/scripts/icub-live

The new image is going to be tested on Blacky with the help of @xEnVrE and @davidelasagna

pattacini, 2021-04-22:
Awesome! Thanks heaps @mbrunettini 👍🏻

closed 2021-04-22 by pattacini