mesh-iit/documentation #118: Improve Bluetooth instruction in the iCubOS section

mbrunettini, 2021-05-18:
As reported by @violadelbono and @Uboldi80 in the icub-tech-iit/tickets#1300 after the merge of https://github.com/icub-tech-iit/documentation/pull/100 the bluetoot instruction [here](https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/bluetooth/) are not clear and can be improved.

Action points are
1. @S-Dafarra organize and hands-on session with @mbrunettini @davidelasagna @violadelbono and @Uboldi80
2. (after collecting feedbacks) @S-Dafarra to improve the guide
3. @mbrunettini to check the resulting PR before merging

violadelbono, 2021-05-18:
@S-Dafarra Thank you! Good for tomorrow. We can use the new robot in production

violadelbono, 2021-05-19:
Ciao @S-Dafarra 

Here the error to add in the documentation:

```
Attempting to connect to 00:06:66:61:FC:04
[CHG] Device 00:06:66:61:FC:04 Connected: yes
[CHG] Device 00:06:66:61:FC:04 ServicesResolved: yes
Failed to connect: org.bluez.Error.NotAvailable
```

S-Dafarra, 2021-05-19:
Here the improvements I noted:
 - Avoid to duplicate the initial Bluetooth commands in the two sections. Simply refer to the previous section.
 - Add some more details on why it is not possible to connect as a classic device, adding the error of the comment above
 - Explicitly state that the initial ``rfcomm connect`` is to check that the connection works
 - In the ``rfcomm bind`` command I forgot the greeny BCB MAC address
 - Explicitly suggest to run the script before starting the service
 - The script should use ``"`` before and after the MAC address
 - Suggest to check if the ``/dev/rfcomm0`` exists

Anything else you would like me to change?

closed 2021-05-20 by mbrunettini