mesh-iit/documentation #142: Clarify what "real" icub server means

traversaro, 2021-07-14:


traversaro, 2021-07-14:
Notes from discussion with @davidelasagna and @Uboldi80 : 
* srv identified both a machine without yarp (when icub is delivered with rack) and laptop machine with yarp and yarpmotorgui (when icub is delivered only with laptop)

Probably it would be worth to clarify naming.

traversaro, 2021-07-14:
A naming seems to be proposed in https://icub-tech-iit.github.io/documentation/icub_operating_systems/#os-for-other-machines-in-the-icub-enviroment (difference between "iCub Server"  and "iCub Server Laptop") but I seems that it is not aligned with the lables assigned to the machines.

traversaro, 2021-07-15:
We had a quick f2f meeting on this on 2021/07/15 with:
* @mbrunettini
* @davidelasagna
* @Uboldi80 
* @Fabrizio69 

It was agreed that to improve the current situation, we will try to unify the naming in this way (see https://icub-tech-iit.github.io/documentation/icub_operating_systems/ for the existing documentation) :
* The machine that only handles the Network Gateway, NFS, DHCP (former "Server" or "Real Server") is going to be called **iCub Dedicated Server**, to highlight that its only role is handling this IT-related role.
* The term for a generic machine (such as a laptop or a workstation) connected to the robot network with YARP and related software installed is going to be **iCub Console**.
* For deployment of iCub in which we have only the iCub and a laptop, there is tipically a single laptop that handls Network Gateway and also has YARP and related software installed (former "iCub Server Laptop"). This kind of machine is going to be called **iCub Console Server**, to highlight the fact that it is related to both roles.

We use this issue to track the necessary changes in the documentation to use this new terminology.

Other notes from the meeting that are related to documentation:
* We current have a quite nice documentation describing the network setup of the robot in https://icub-tech-iit.github.io/documentation/icub_operating_systems/icubos/networking/ . However, it describes the configuration with a **iCub Dedicated Server**, not the one with a **iCub Console Server** that instead is (according to @Fabrizio69) the configuration that we are actually using more for iCub that are sent outside IIT in the last years. For this reason, it could make sense to expand the network configuration to also describe the network configuration with a **iCub Console Server** .

mbrunettini, 2021-07-16:
First of all, I want to improve the machine roles description in [this page](https://icub-tech-iit.github.io/documentation/icub_operating_systems/), something like the following:

> Once you customized your machine, you can proceed with further setup for special machines, such as:
> 
> - [_The iCub Dedicated Server_](./other-machines/icub-server-os.md) - the machine that works as _network infrastructure server_ (for Network gateway, NTP, NFS, DHCP and DNS services) only, this machine is not supposed to be used to run your software (i.e. you can't install YARP here), but acts as backend server for all networking services
> - [_The iCub Console Server_](./other-machines/icub-server-laptop.md) - in the minimal iCub setup configuration there is only the robot and a single machine, usually a laptop, acting both as (limited) _network infrastructure server_ (for Network Gateway, ntp and NFS serviceas) as well as _robot console_ were you can run your software (i.e. YARP is installed here)

What do you think?

traversaro, 2021-07-16:
Seems nice for me.

closed 2021-07-16 by pattacini

traversaro, 2021-07-16:
Just to clarify, I think that if @Uboldi80 has any comment on the modification it would be more then welcome even now that the PR has been already merged.

Uboldi80, 2021-07-16:
Nice form me, thanks Silvio.