mesh-iit/documentation #143: Clarify what "skip in case of iCub Server" means in https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/generic-machine/#icub-software-repository-and-common-packages

traversaro, 2021-07-14:
It is not clear what "skip in case of iCub Server"  means there. With @Uboldi80 we were setting up a iCub Server Laptop and we were not sure if we had to skip it.

traversaro, 2021-07-14:
The link prints https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-laptop/#software-repositories .

traversaro, 2021-07-14:
This probably requires https://github.com/icub-tech-iit/documentation/issues/142 to be solved before working on this.

pattacini, 2021-07-14:
cc @mbrunettini 
As @traversaro said, to be addressed along with #142.

traversaro, 2021-07-15:
As https://github.com/icub-tech-iit/documentation/issues/142 evolved a bit, now this is issue is basically a subset of https://github.com/icub-tech-iit/documentation/issues/142, so I think we can close it.

closed 2021-07-15 by traversaro