mesh-iit/documentation #151: SwiftGPU link is broken

traversaro, 2021-07-15:
SwiftGPU link in https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/cuda-workstation/#install-siftgpu-library-and-its-dependancies is broken. Not sure is that dependency is still neeed or not. @xEnVrE do you know anything about this?

pattacini, 2021-07-15:
Just discussed this w/ @vtikha and we may get rid of it as the latest IOL detection layer makes use of Caffe instead.
@xenvre may give more insights on this anyway.

xEnVrE, 2021-07-15:
It is used here also https://github.com/robotology/stereo-vision/blob/master/modules/SFM/CMakeLists.txt.

However, seems that the [root](https://github.com/robotology/stereo-vision/blob/00b18ba214463db9037ea915e25a454b9bf954ec/CMakeLists.txt#L21) CMakeLists.txt is ignoring it anyways.

Btw, never had the necessity to compile `stereo-vision` with support for SiftGPU in the past. I think it can be removed as well.

PR #152 (2021-07-15): Update cuda-workstation.md
Fix #151.
closed 2021-07-15 by pattacini