mesh-iit/documentation #161: Migrate old wiki page on the iCub system setup

pattacini, 2021-07-30:
It'd be nice to move [How to set up the iCub workstation from scratch](https://github.com/robotology/robotology.github.io/wiki/How-to-set-up-the-iCub-workstation-from-scratch) here and check whether there might be overlapping with already existing resources.

That documentation needs to be revised anyhow as we now rely on the superbuild and on a new standard setup (e.g. `pc104` is now `icub-head`). 

Finally, the original wiki page should be removed.

pattacini, 2021-07-30:
@sgiraz @mfussi66, you could start having a look to get familiar ahead of the robot hands-on.

pattacini, 2021-08-03:
@sgiraz, @mfussi66, I'll be most likely dealing with the migration.
Don't worry.

pattacini, 2021-08-04:
@davidetome volunteered for this ❤️ 
We may also leap at the chance to integrate some other internal documentation.

davidetome, 2021-09-16:
Today we had a meeting and we decided to not proceed w/ this activity seen the fact that most of the info contained in the document is out of date.
Anyway, @mbrunettini and @davidelasagna will open a ticket for writing down new documentation for internal use, this doc should be completed/maintained in the future by newcomers.

I think we can close this issue then.

cc @pattacini @vtikha

closed 2021-09-16 by pattacini

traversaro, 2021-09-16:
> Today we had a meeting and we decided to not proceed w/ this activity seen the fact that most of the info contained in the document is out of date.

If that document is out of date, can we just remove it from the wiki. If we ever need that, we can always recover it from the wiki history, but if we leave it as it is people will find it via Google search and will assume that it is valid information.

pattacini, 2021-09-16:
Sure, I was going to do right that 👍🏻