mesh-iit/documentation #174: Document iCub 3 joint limits

pattacini, 2021-09-09:
See https://github.com/robotology/icub-models-generator/pull/202#discussion_r624434849.

cc @traversaro

traversaro, 2021-09-09:
The documentation in https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/ (generated from https://github.com/icub-tech-iit/documentation/blob/a828191b77c30df153c9dcb2156b1d8cce9ec9a0/docs/icub3/icub3-joints.md) has some information on hardware limits, while software limits are missing from several joints. Documenting hardware and software limits is important as this information is encoded in several different points of the software infrastructure, including:
* Hardware Limits: contained in https://github.com/robotology/icub-models-generator/blob/master/simmechanics/data/icub3/ICUB_3_joint_all_parameters.csv that is feeded in the URDF and SDF files
* Software Limits: contained in Gazebo yarp controlboard configuration files's LIMITS section https://github.com/robotology/icub-models-generator/blob/v1.21.1/simmechanics/data/icub3/conf/gazebo_icub_left_arm_no_hand_for_no_hand_model.ini#L70
* Software Limits: contained in robot-configuration files https://github.com/robotology/robots-configuration/blob/v1.21.0/iCubGenova09/hardware/motorControl/left_arm-eb1-j0_1-mc.xml#L14

I could have missed some other points in which limits information is hardcoded, but even just those one highlight the importance of clearly documenting this limits. In general, as a starting point we can simply just take the values that are present in these configuration files and add them to the documentation. If those values are wrong, we can always change them later. 

There are already some values that are clearly inconsistent. For example, for the shoulder roll we had a discussion in https://github.com/robotology/icub-models-generator/pull/202#discussion_r624434849 that the **software** upper limit is 163, but the documentation at the moment reports that the **hardware** upper limit is 150, see https://github.com/icub-tech-iit/documentation/blame/1ddb4dee540033d3ff10a48c5ab388d618e5021e/docs/icub3/icub3-joints.md#L125 so probably also limits already present in the document may need to be reviewed.

traversaro, 2021-09-20:
Example of issue in which this would be useful: https://github.com/robotology/icub-models/issues/104#issuecomment-922744739 .

pattacini, 2023-07-30:
Same as https://github.com/icub-tech-iit/documentation/issues/175#issuecomment-1657210237.

cc @traversaro

closed 2023-07-31 by pattacini