mesh-iit/documentation #175: Update iCub 3 joint documentation to document joint names used in URDF files and configuration parameters

traversaro, 2021-09-09:
In several points of the software infrastructure, joint are identified with strings, for example the joint `r_shoulder_roll` for iCub3 in 👍 * https://github.com/robotology/robots-configuration/blob/v1.21.0/iCubGenova09/hardware/mechanicals/left_arm-eb1-j0_1-mec.xml#L9
* https://github.com/robotology/icub-models-generator/blob/v1.21.1/simmechanics/data/icub3/ICUB_3_joint_all_parameters.csv#L6

However, the documentation in https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/ does not report this identifier for the joints, creating confusion as the mapping between is not clear.

pattacini, 2023-07-30:
I would close this issue as not planned, @traversaro.
iCub3 is now iRonCub3 and differs significantly from its original mechs.
Also, I would focus effort on ergoCub.

Do you agree?

traversaro, 2023-07-31:
Ok!

closed 2023-07-31 by pattacini