mesh-iit/documentation #176: Document/clarify existence of iCub version 1.7

traversaro, 2021-09-13:
The documentation on iCub arm kinematics report slighly different values for the "1.7" arm, see https://icub-tech-iit.github.io/documentation/icub_kinematics/icub-forward-kinematics/icub-forward-kinematics-arms/ . However, in https://icub-tech-iit.github.io/documentation/icub_versions/ there is no documentation for iCub 1.7 .

maggia80, 2021-09-16:
I didn't know the existence of arm 1.7.

pattacini, 2021-09-16:
We do have it, actually.
Our codebase mirrors it: see for example https://github.com/robotology/icub-main/blob/master/src/libraries/iKin/src/iKinFwd.cpp#L1699.

traversaro, 2021-09-16:
According to robots-configuration files, the following robots have v1.7 arms (https://github.com/robotology/robots-configuration/search?q=left_v1.7):
* `iCubParis01`
* `iCubHertfordshire01`
* `iCubGenova03`

commit 1b846a03 (2023-07-31): fix #176
closed 2023-07-31 by pattacini