mesh-iit/documentation #184: Update motor type and harmonic drive details on iCub 3 joint documentation

divyashah, 2021-11-22:
I observed that the details for a few of the joints in [iCub 3 joints specs](https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/), in particular `Motor type` and `Harmonic drive`, are not up to date with the latest CAD model. I believe the documentation has not been updated after some of the latest upgrades on iCub 3. 

I noticed mainly the following inconsistencies: 
- [Joint 12, 18 - left, right foot roll](https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/#joint-12-18-left-right-foot-roll): `Motor type` should be updated from ~MOOG_C2900575~ to MOOG_C2900576 and `Harmonic drive` should be updated from ~CSD-17-160-2A~ to CSD-17-100-2A. 
- [Joint 7, 13 - left, right hip pitch](https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/#joint-7-13-left-right-hip-pitch): `Harmonic drive` should be updated from ~CSD-20-100-2UH~ to CSD-25-100-2UH.
- [Upperbody Joints specs](https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/#upperbody-joints-specs): For all of the upperbody joints, namely `yaw`, `shoulder_pitch`, `shoulder_roll`, `upperarm_yaw` and `elbow`, the details on `Harmonic drive` are missing. 

How shall we proceed in order to update the [iCub 3 joints specs](https://icub-tech-iit.github.io/documentation/icub3/icub3-joints/) documentation ??  
cc @fiorisi @pattacini

pattacini, 2021-11-22:
>How shall we proceed in order to update the iCub 3 joints specs documentation ??

Hi @divyashah 

PR's are more than welcome.
Just calling for @salvi-mattia review as well.

pattacini, 2023-07-30:
Hi @divyashah 

Would you be willing to contribute on this?

divyashah, 2023-07-31:
> Hi @divyashah
> 
> Would you be willing to contribute on this?

Hi @pattacini 

Yes, I can contribute on this. 
However, should I update now modify the page on iCub3 itself or shall we create a new page for ergoCub1 ?

pattacini, 2023-07-31:
> Yes, I can contribute on this.

Thanks heaps @divyashah! Very much appreciated.

>However, should I update now modify the page on iCub3 itself or shall we create a new page for ergoCub1 ?

My hunch is that we should move to ergoCub, whose joint specs are missing in the documentation. Such a page is very overdue ⏰

I reckon that to start with, it'd be sufficient to create a page similar to that of iCub 3 but without images, to keep things simple.

What do you think?

divyashah, 2023-07-31:
> My hunch is that we should move to ergoCub, whose joint specs are missing in the documentation. Such a page is very overdue ⏰
> 
> I reckon that to start with, it'd be sufficient to create a page similar to that of iCub 3 but without images, to keep things simple.

Yes, sounds good. We can add images later. I will start preparing a draft on this.

pattacini, 2023-07-31:
Superb!

If it's your first-time contribution, I'd suggest you rely on this workflow:
- https://github.com/icub-tech-iit/documentation#-github-pages-on-forks

pattacini, 2023-07-31:
Hi @divyashah 
While addressing this task, please account for https://github.com/icub-tech-iit/documentation/issues/175#issue-991946032.

cc @traversaro

closed 2023-08-09 by pattacini