mesh-iit/documentation #192: Feedback from installation of new iCubOS image on iCubGenova04

traversaro, 2021-12-17:
On 2021/12/17, the iCubGenova04 icub-head was updated, following the docs in https://github.com/icub-tech-iit/documentation/blob/0db3f48fe41a71dfcedf93d989cc32e71f390d1e/docs/icub_operating_systems/icubos/installation-from-image.md . This is a list of issue/small annoyances experienced:
* From https://icub-tech-iit.github.io/documentation/icub_operating_systems/download/#download-table, it is not clear which version of iCub needs iCubOS.
* Among the required material, a DisplayPort cable was not listed (it is kind of implied from the monitor)
* It is not clear how to access the BIOS

traversaro, 2021-12-17:
> It is not clear how to access the BIOS

This happened as for some reason the usual starting screen did not appear. However, we were able to understand that the key to press was the `DEL` one as both (See https://icub-tech-iit.github.io/documentation/icub_cpu_boards/icub_cpu_boards/) the `conga-TC87/i7-4650U` and `COMe-cKL6i7-7600` manuals (see page 74 of https://www.congatec.com/fileadmin/user_upload/Documents/Manual/TU87.pdf and page 61 of https://www.kontron.com/download/download?filename=/downloads/manuals/c/come-ckl6_user-guide_rev-1.5_2020-07-23.pdf&product=146491) reported that `DEL` was the correct key to use to enter the BIOS.

traversaro, 2021-12-17:
fyi @HosameldinMohamed @gabrielenava

pattacini, 2021-12-17:
cc @mbrunettini

traversaro, 2021-12-17:
We also have some other problems that we are not able to fix, but probably @HosameldinMohamed will open an issue in https://github.com/robotology/community/discussions to ask directly support for that as I am not able to fix the problem that he is experiencing.

pattacini, 2021-12-17:
Maybe https://github.com/icub-tech-iit/tickets is a better place?

traversaro, 2021-12-17:
> Maybe our internal tickets system is a better place?

Ok for me.

pattacini, 2023-07-30:
Hi @traversaro 

Is this still an issue?