mesh-iit/documentation #195: Not documented installation of ycm-cmake-modules in iCubOS 8.0 relased on 2021/06/04?

traversaro, 2022-01-31:
In the last days, both @Uboldi80 and @HosameldinMohamed reported this error when compiling the robotology-superbuild on a brand new iCubOS:
~~~
CMake Error at CMakeLists.txt:33 (find_package):
  Could not find a configuration file for package "YCM" that is compatible
  with requested version "0.13.0".

  The following configuration files were considered but not accepted:

    /usr/share/cmake/YCM/YCMConfig.cmake, version: 0.12.2
~~~

The problem is that YARP finds an old YCM installed in the system, and so it does not proceed as it requires YCM 0.13 (that would be downloaded by the superbuild if a YCM was not found in the system). 
The solution is to just `sudo apt update` and `sudo apt upgrade` that updates YCM to 0.13 . 

However, as this problem already affected two users in a few days, probably we should specify that `sudo apt update` and `sudo apt upgrade` is necessary, either after the iCubOS from image installation, or before the software installation?

pattacini, 2022-01-31:
cc @mbrunettini

pattacini, 2022-02-04:
I think it's faster here if someone provides a PR that we can discuss/comment anyhow.

gsisinna, 2022-03-02:
We will insert a first step in the documentation with:

`sudo apt update`
`sudo apt upgrade`

This will install the last version (0.13) of YCM.

Usually this the first step with a new Linux system, but I can modify the documentation and specify that.

pattacini, 2022-03-02:
Good 👍🏻