mesh-iit/documentation #246: Document the coupling law of Hand MK5

pattacini, 2023-01-13:
We can publish this piece of info in https://icub-tech-iit.github.io/documentation/hands/.

cc @traversaro @fiorisi

mfussi66, 2023-01-13:
At the moment we have the following documents:
Description of the coupling law: 
https://github.com/icub-tech-iit/dev/blob/master/projects/icub-hand-Mk3/calculations/FingerkinV2.docx

Definition of kinematic quantities in this table:
https://istitutoitalianotecnologia.sharepoint.com/:x:/r/sites/icub-facility/_layouts/15/Doc.aspx?sourcedoc=%7BACD707A3-EE3D-469F-BBCE-D47664B4F432%7D&file=Fingerkin_hand_mk4.xlsx&action=default&mobileredirect=true

I'll see if I can find other info.

fiorisi, 2023-01-13:
> I'll see if I can find other info.

Probably @ale-git can point you to additional documentation.

traversaro, 2023-01-13:
> At the moment we have the following documents: [Finger_handMk4_kinematics.zip](https://github.com/icub-tech-iit/documentation/files/10411250/Finger_handMk4_kinematics.zip)
> 
>     * the docx contains the kinematic coupling
> 
>     * the xlsx contains the kinematic quantities
> 
> 
> I'll see if I can find other info.

Just to clarify, does the document refer to mk4 or mk5?

pattacini, 2023-01-13:
>Just to clarify, does the document refer to mk4 or mk5?

Fingers didn't undergo substantial modifications from mk4 to mk5, except for the loss of the thumb in-place rotation.

ale-git, 2023-01-13:
> > I'll see if I can find other info.
> 
> Probably @ale-git can point you to additional documentation.

Yes, let me dig into my Team Dev old issues...

ale-git, 2023-01-19:
The document posted by Silvio is my most recent, anyway in this issue there is some history of the development.
https://github.com/icub-tech-iit/dev/issues/42

Here it is an implementation for the gazebo yarp plugin:

https://github.com/robotology/gazebo-yarp-plugins/blob/fe7a39e932f35488ee292ee77f7d9853b4ac57a6/plugins/controlboard/src/ControlBoardDriverCoupling.cpp#L1136

Nicogene, 2023-02-03:
Documentation ported to markdown and added to the online documentation (https://github.com/icub-tech-iit/documentation/pull/249).

For the future maybe it is better to write this documentation directly in markdown in order to ease the insertion in the official documentation.

closed 2023-02-04 by pattacini