mesh-iit/documentation #250: Document how to access the skin data in iCub robots

xEnVrE, 2023-02-22:
Several information on how to read the low-level and high-level data from the skin sensors in iCub 2.x robots is available in the [Tactile sensors (aka Skin)](http://wiki.icub.org/wiki/Tactile_sensors_(aka_Skin)) page from the "old" iCub wiki. 

A section with these information is missing in the `iCub Tech Docs` website, at least I could not find it. It could probably contain an extract of what is written in the old wiki, not sure if all the information there are still up to date. 

As regards iCub 3, I think these information would also apply to the hand palm and fingertips of that robot.

pattacini, 2023-02-23:
I'll be on it on PI14-sprint03 (i.e., Mar 6 – Mar 17).

pattacini, 2023-03-09:
Started off at https://github.com/icub-tech-iit/documentation/tree/feat/tactile-sensors.

I'll first port the info as it is and then I'll try to revise it.

pattacini, 2023-03-12:
The PR is open.

I would need to review mostly the information highlighted in the old wiki page.

| 🔘 Click to download the PDF |
| :---: |
| [![][img]][pdf] |

[img]: https://user-images.githubusercontent.com/3738070/224562349-313bf572-2526-4531-b93a-847c39491e4a.png
[pdf]: https://github.com/icub-tech-iit/documentation/files/10951567/tactile_data.pdf