mesh-iit/documentation #271: Document convention for ports/devices exposed in iCub-like robots

traversaro, 2023-05-10:
I wanted to work on https://github.com/robotology/robots-configuration/issues/485, but I am missing even a rough guideline related to naming. I will write one draft here, then we can migrate this to proper documentation when it is ready.

## iCub YARP network wrapper server conventions

iCub exposes its functionalities (sensors and actuators) via YARP devices called [YARP's Network Wrapper Servers (NWSs)](https://www.yarp.it/latest//group__nws__and__nwc__architecture.html) that publish information over YARP ports. For historical reasons, for each type of functionality  such as "controlboards" (i.e measure of joint-related sensors and capability of setting setpoints of low-level control loops), inertial measurement units, force torque sensors, multiple devices are used, one for each iCub's "part". 

The "parts" of iCub are:
* `head`
* `left_arm` 
* `right_arm`
* `torso`
* `left_leg`
* `right_leg`

Not all iCubs have all its six parts. There are iCub variants that are composed only by the `head`, or iCub variants that are composed only by the legs. In some contexts also a special part `alljoints` is defined and used, that is just a part that actually represents all the differents part of the iCub combined.

In the following, we will list the convention used for each type of functionality.

In the following, we will refer also to the `robotPortPrefix`, that is a string that can have multiple values:
* `icub` for physical iCub robots
* `icubSim` for simulated iCub robots
* `ergoCub` for physical ergoCub robots
* `ergoCubSim` for simulated ergoCub robots

### ControlBoard

The `ControlBoard` functionality is exposed via the [`controlBoard_nws_yarp`](https://www.yarp.it/latest/classControlBoard__nws__yarp.html) device. The `name`  parameter passed to this device is: `/<robotPortPrefix>/<partName>`. 

This means that for each part there will be a `controlBoard_nws_yarp` device that will open the following YARP ports:
* `/<robotPortPrefix>/<partName>/state:o` : that publishes encoder information for the part
* `/<robotPortPrefix>/<partName>/stateExt:o` : that publishes different joint-level sensors information for the part, using the structure defined in https://github.com/robotology/yarp/blob/master/src/libYARP_dev/src/idl/stateExt.thrift
* `/<robotPortPrefix>/<partName>/rpc:i` : that expose several information related to the part via a YARP RPC port
* `/<robotPortPrefix>/<partName>/command:i` : that takes in input references for the low-level control loops

**Note: these YARP ports are not meant to be accessed directly, but should be accessed instead via the [`remote_controlboard`](https://www.yarp.it/latest//classRemoteControlBoard.html) device.**

### Inertial Measurements Units (IMUs)

Inertial Measurements Units functionalities are exposed via the [`multipleanalogsensorsserver`](https://www.yarp.it/latest//classMultipleAnalogSensorsServer.html) YARP device. 
The `name`  parameter passed to this device is: `/<robotPortPrefix>/<partName>/inertials`. 

This means that for each part there will be a `multipleanalogsensorsserver` device that will open the following YARP ports:
* `/<robotPortPrefix>/<partName>/inertials/measures:o` : that publishes sensors information for the part, using the structure defined in https://github.com/robotology/yarp/blob/master/src/devices/multipleAnalogSensorsMsgs/multipleAnalogSensorsSerializations.thrift
* `/<robotPortPrefix>/<partName>/inertials/rpc:o` : that expose several information related to the part via a YARP RPC port

**Note: these YARP ports are not meant to be accessed directly, but should be accessed instead via the [`multipleanalogsensorsclient`](https://www.yarp.it/git-master/classMultipleAnalogSensorsClient.html) device.**

This device will expose sensors related to the following [Multiple Analog Sensors Interfaces](https://www.yarp.it/latest/group__dev__iface__multiple__analog.html):
* [`yarp::dev::IOrientationSensors`](https://www.yarp.it/latest/classyarp_1_1dev_1_1IOrientationSensors.html)
* [`yarp::dev::IThreeAxisGyroscopes`](https://www.yarp.it/git-master/classyarp_1_1dev_1_1IThreeAxisGyroscopes.html)
* [`yarp::dev::IThreeAxisLinearAccelerometers`](https://www.yarp.it/git-master/classyarp_1_1dev_1_1IThreeAxisLinearAccelerometers.html)
* [`yarp::dev::IThreeAxisMagnetometers`](https://www.yarp.it/git-master/classyarp_1_1dev_1_1IThreeAxisMagnetometers.html)

### Force-Torque Sensors

Force-Torque Sensors functionalities are exposed via the [`multipleanalogsensorsserver`](https://www.yarp.it/latest//classMultipleAnalogSensorsServer.html) YARP device. 
The `name`  parameter passed to this device is: `/<robotPortPrefix>/<partName>/FT`. 

This means that for each part there will be a `multipleanalogsensorsserver` device that will open the following YARP ports:
* `/<robotPortPrefix>/<partName>/FT/measures:o` : that publishes sensors information for the part, using the structure defined in https://github.com/robotology/yarp/blob/master/src/devices/multipleAnalogSensorsMsgs/multipleAnalogSensorsSerializations.thrift
* `/<robotPortPrefix>/<partName>/FT/rpc:o` : that expose several information related to the part via a YARP RPC port

**Note: these YARP ports are not meant to be accessed directly, but should be accessed instead via the [`multipleanalogsensorsclient`](https://www.yarp.it/git-master/classMultipleAnalogSensorsClient.html) device.**

This device will expose sensors related to the following [Multiple Analog Sensors Interfaces](https://www.yarp.it/latest/group__dev__iface__multiple__analog.html):
* [`yarp::dev::ISixAxisForceTorqueSensors`](https://www.yarp.it/git-master/classyarp_1_1dev_1_1ISixAxisForceTorqueSensors.html)
* [`yarp::dev::ITemperatureSensors`](https://www.yarp.it/git-master/classyarp_1_1dev_1_1ITemperatureSensors.html)

pattacini, 2023-05-10:
Super!

cc @Nicogene

traversaro, 2023-05-17:
I have a bit of doubt on the "Inertial Measurements Units (IMUs)" part. I proposed the scheme for the IMU to be `/<robotPortPrefix>/<partName>/inertials` , but apparently the scheme used in ergocub real robot is `/<robotPortPrefix>/<partName>/imu`, not sure which one we want to use.

pattacini, 2023-05-17:
I've got no particular preference.
We can maybe pick the convention followed by the majority of the robots and then adjust the rest.

@Nicogene @davidetome any comment?

davidetome, 2023-05-17:
> I've got no particular preference. We can maybe pick the convention followed by the majority of the robots and then adjust the rest.
> 
> @Nicogene @davidetome any comment?

Sounds good to me 👍🏻

Nicogene, 2023-05-17:
> I've got no particular preference. We can maybe pick the convention followed by the majority of the robots and then adjust the rest.
> 
> @Nicogene @davidetome any comment?

I agree 👍🏻

pattacini, 2023-07-19:
Work done via:
- https://github.com/icub-tech-iit/documentation/pull/292

Let me close this issue and put a reminder for @traversaro's final review.

/remind 01 August, please, @traversaro have a look at the documentation in this respect.

closed 2023-07-19 by pattacini