mesh-iit/documentation #307: Update coupling parameters for Hand MK5.1

xEnVrE, 2023-09-11:
Today, while reviewing the newly opened PR https://github.com/icub-tech-iit/ergocub-software/pull/169, I had the chance to compare the [documented](https://icub-tech-iit.github.io/documentation/hands/hands_mk5_coupling/#mk51-coupling-variables) parameters of the coupling law for the MK5 **.1** hand and a table that can be found in https://github.com/icub-tech-iit/ergocub-design-lowerarm/issues/157#issuecomment-1674984233.

It seems that there are discrepancies for the following paremeters.

**Note:** green and yellow highlighed text is just coming from the screenshot of the "table" indicated above. I am not highlighting anything. 

Moreover please take into account that for each "table" entry the order of numbers is "thumb", "index" and "pinky" (as index == middle == ring) while each "documentation" entry report all the 5 parameters in the order "thumb", "index", "middle", "ring", "pinky".

# Parameter $q_{0off}$

### table
[image]

### documentation

[image]

# Parameter $k$

### table

[image]

### documentation

[image]

# Parameter $d$

### table

[image]

### documentation

[image]

# Parameter $l$

### table
[image]

### documentation
[image]

# Parameter $b$

### table

[image]

### documentation
[image]

# Parameter $s$

### table
[image]

### documentation
[image]

# Parameter $t$

### table
[image]

### documentation
[image]

# Parameter $f$

### table

[image]

### documentation

[image]

# Paramter $r$

### table

[image]

### documentation

[image]

cc @pattacini

pattacini, 2023-09-11:
cc @fiorisi @Lawproto

Lawproto, 2023-09-12:
@xEnVrE a small disclaimer. The excel file only served me to update the documentation, it should not be considered official. The values you screenshotted were reported to me from @ale-git as a double-check, are in `m` instead of `mm` and I do not exactly know how they have been calculated when there is some difference with the one I measured via CAD (yellow highlight). In general, I considered the difference small enough and I put the value coming from the CAD inside the documentation.

However, I just noticed that for the single letter length parameters I accidentally copied the Mk5.0 parameters instead of putting the correct one. I thank you for pointing this out, I will proceed with an update via PR.

xEnVrE, 2023-09-12:
Thanks for the clarification @Lawproto. 

Indeed I though they were not super official but the parameters were the same as `Mk 5.0`, as you said, hence I though it was important to notify that as it seemed very strange to me. 

Moreover while I was conducting experiments on Gazebo, in order to study the length of the fingers, I have also asked some numbers to @ale-git and they were very different from those of the documentation, hence I was kinda sure that something was wrong.

Lawproto, 2023-09-12:
PR opened. 
https://github.com/icub-tech-iit/documentation/pull/312

closed 2023-09-13 by Nicogene