mesh-iit/documentation #324: Updating Robot PC Setup Documentation to Prevent Duplicate Environment Variable Loading

GiulioRomualdi, 2023-10-31:
With @lrapetti @mebbaid and @traversaro and I noticed that the variable exported in `.bashrc_ergoCub` were loaded twice when a new terminal is opened. 

Upon further investigation, it was discovered that the following code:
```bash
  if [ "$HOME" != "" ]; then                                                                                                                                                                               
      ICUBRC_FILE="${HOME}/.bashrc_ergoCub"                                                                                                                                                                
  else                                                                                                                                                                                                     
      ICUBRC_FILE="/home/icub/.bashrc_ergoCub"                                                                                                                                                             
  fi                                                                                                                                                                                                       
  if [ -f "$ICUBRC_FILE" ]; then                                                                                                                                                                           
      source $ICUBRC_FILE                                                                                                                                                                                  
  fi  
```
was placed above the following block of code in `.bashrc`:
```bash
# If not running interactively, don't do anything                                                    
case $- in                                                                                           
    *i*) ;;                                                                                          
      *) return;;                                                                                    
esac 
```

so when the gui session was loading the `bash` it was exporting all the environment variables in `.bashrc_ergoCub`

To fix this we modified the `.bashrc` adding the following lines at the beginning of the file

```bash
if [[ $- == *i* ]] || [[ -n "$SSH_CLIENT" ]] || [[ -n "$SSH_TTY" ]]; then                                                                                                                                  
  #Load the ergoCub custom bashrc                                                                                                                                                                          
  if [ "$HOME" != "" ]; then                                                                                                                                                                               
      ICUBRC_FILE="${HOME}/.bashrc_ergoCub"                                                                                                                                                                
  else                                                                                                                                                                                                     
      ICUBRC_FILE="/home/icub/.bashrc_ergoCub"                                                                                                                                                             
  fi                                                                                                                                                                                                       
  if [ -f "$ICUBRC_FILE" ]; then                                                                                                                                                                           
      source $ICUBRC_FILE                                                                                                                                                                                  
  fi                                                                                                                                                                                                       
fi   
```

In this code snippet, `[[ $- == *i* ]]` checks if the $- variable contains the i character, which indicates an interactive shell. `[[ -n "$SSH_CLIENT" ]]` checks if the SSH_CLIENT environment variable is non-empty, which indicates an SSH session. Similarly, `[[ -n "$SSH_TTY" ]]` checks if the `SSH_TTY` environment variable is non-empty, which also indicates an SSH session.

If either of these conditions is true, the code inside the if block is executed. You can replace the echo command with the command you want to run.

To ensure others do not encounter this issue, we should update the documentation on how to set up a robot PC.

lrapetti, 2023-10-31:
The existing documentation that we might need to update can be found in https://github.com/icub-tech-iit/documentation/blob/5d919960f2d71e25d845b1eb4b899ed0c389ae43/docs/icub_operating_systems/pc104/the-icub-live-details.md?plain=1#L302-L351.

pattacini, 2023-10-31:
Hi @GiulioRomualdi @lrapetti 
Thanks for pointing it out 👍🏻 
Would you be willing to provide a PR?

cc'ing @davidelasagna for any possible update to be made to the images.

traversaro, 2023-10-31:
Thanks @GiulioRomualdi ! 

> The existing documentation that we might need to update can be found in
> 
> https://github.com/icub-tech-iit/documentation/blob/5d919960f2d71e25d845b1eb4b899ed0c389ae43/docs/icub_operating_systems/pc104/the-icub-live-details.md?plain=1#L302-L351
> .

While for consistency we may want to update also that documentation, as the `icub-head` or similar systems are not tipically used via a graphical session, that is not a big problem. I think the most important place to update is the docs in https://github.com/icub-tech-iit/documentation/blob/master/docs/icub_operating_systems/icubos/user-env.md#how-to-setup-the-enviroment-properly .

lrapetti, 2023-10-31:
> While for consistency we may want to update also that documentation, as the `icub-head` or similar systems are not tipically used via a graphical session, that is not a big problem. I think the most [...]

Yes the one you are pointing is the actual documentation to be updated. In the one I was pointing there was not specified the location w.r.t. the interactive configuration.

In addition, I think a lot of the people (e.g. me) are not aware that `yarpmanager` commands (e.g. running the `yarpserver`) on their personal laptop is not working because of the place in which we call [robotology-superbuild `setup.sh`](https://github.com/robotology/robotology-superbuild#configure-your-environment) w.r.t. the run interactively check (@traversaro maybe we can add this to the robotology-superbuild documentation).

GiulioRomualdi, 2023-10-31:
Hi @pattacini here you have: https://github.com/icub-tech-iit/documentation/pull/325

pattacini, 2023-10-31:
Thanks heaps @GiulioRomualdi 👍🏻

closed 2023-11-03 by pattacini