mesh-iit/documentation #334: Extend the documentation of wrist math

Nicogene, 2024-02-15:
Follow-up of:
- https://github.com/robotology/icub-firmware-models/pull/82#pullrequestreview-1882075579

pattacini, 2024-02-16:
Hi @ale-git

While extending the documentation, we can also describe the situation when the neck system can get stuck at startup.

pattacini, 2024-03-04:
We should also address the problem of disambiguation of the startup configuration.
See https://github.com/icub-tech-iit/knowledge-base/discussions/17.

cc @ale-git @Nicogene

ale-git, 2024-03-19:
There is an important fact to consider: the `motors2ypr` solver module never fails if the gap between each couple of arms is less than 180 deg. I have checked this fact by spanning all the allowed arm configurations under this condition:
[image]

ale-git, 2024-03-19:
The same check in the `ypr2motors` module should prevent us to reach such singularities. We could add some safety threshold in order to prevent unallowed configurations also in case of overshooting.

The motors2ypr solver works in this way:

- the gradient descent step is initialised to 0.01, and the ypr configuration is initialised with that calculated in the previous control cycle
- if the algorithm can't converge (singularity) then it performs backtraking to the last good ypr configuration, and divides the step by 2, and continues
- if the step runs below 0.001 then we have reinitialization to (0,0,0) know position, and step = 0.01

ale-git, 2024-03-19:
I have updated the documentation with the topics described above.
https://github.com/icub-tech-iit/documentation/pull/343

closed 2024-03-19 by pattacini