mesh-iit/documentation #336: Revert #328

pattacini, 2024-02-27:
After an internal discussion during our board meeting, we decided to revert PR #328.

This is because the convention for the CSYS is limited to the alignment of their axes wrt a resting position of the robot, but doesn't include the directions of rotation. In practical terms, this means that the same joint angles guarantee a specular symmetry between the arms.

Reverting #328 is as simple as clicking a button, although we may need to carry out a careful review.

cc @Lawproto @Nicogene @fiorisi @traversaro

Nicogene, 2024-03-13:
I opened a PR with just the commit reverted:
- https://github.com/icub-tech-iit/documentation/pull/341

But it seems not sufficient, more refinement is needed.

pattacini, 2024-03-25:
A gentle reminder for @Lawproto.
cc @Nicogene

Lawproto, 2024-03-28:
@pattacini there you go:
- https://github.com/icub-tech-iit/documentation/pull/345

closed 2024-03-28 by pattacini