mesh-iit/documentation #365: Document that Debian-based iCubOS and pc104 icub-head are not supported anymore since robotology-distro 2024.11

traversaro, 2024-10-30:
Since https://github.com/robotology/robotology-superbuild/pull/1547 new versions of the robotology-superbuild do not support Debian 10 and 11 (that is the version available in latest iCubOS). I guess it could be a good idea to mention this in related documentation pages:
* https://icub-tech-iit.github.io/documentation/icub_operating_systems/pc104/icub-live/

Or directly remove this pages to simplify documentation (if somebody needs them they can be found in git history).

traversaro, 2024-10-30:
If I am not wrong, this means dropping support for newer distro for the following non-iceboxed robots (I checked the robots that still use `cfw2can`:
* iCubLondon01
* iCubDijon01
* iCubGrenoble01
* iCubGenova03
* iCubHertfordshire01

Probably we can discuss in person before taking a decision.

pattacini, 2024-10-30:
>Or directly remove this pages to simplify documentation

Right because you spotted old robots not in the icebox, perhaps a better option is rather to add a note in the present documentation.

>Probably we can discuss in person before taking a decision.

Yep 👍🏻

maggia80, 2024-11-05:
iCubGrenoble01 is under upgrade and we will use the COM-EXPRESS removing the CFW2 board

pattacini, 2024-11-05:
iCubHertfordshire02 as well.

pattacini, 2024-11-05:
Relevant robots moved to icebox:
- https://github.com/robotology/robots-configuration/pull/687

cc @traversaro

pattacini, 2024-11-05:
Hi @reafrancesco 

>@maggia80 would it be possible to start similar upgrade procedure for the iCubGenova03?

This is kind of off-topic here. Would you mind reaching out to @maggia80 using other channels?

PR #366 (2024-11-05): Update icub-live.md
Fix #365.
closed 2024-11-05 by pattacini