mesh-iit/documentation #369: Publish the documentation regarding the new ergocub Battery Pack

valegagge, 2024-11-12:
The new ergocub battery pack (Wingst code = 18240) is mounted and working on the ergocub S/N002. 

We need to publish the documentation that provides the information on how to use it from the point of view of the robot user.

andresoll, 2025-01-17:
Hi all,

[here](https://github.com/andresoll/documentation/blob/master/docs/ergoCub_battery_pack/ergoCub_battery_pack.md) the user guide to use the new **ergoCub_battery_pack**.

At the moment still in my documentation fork. I will proceed shortly with a pool request.

@valegagge

pattacini, 2025-01-17:
Great @andresoll 
Remember to try out [generating the website](https://github.com/icub-tech-iit/documentation?tab=readme-ov-file#-github-pages-on-forks) from your fork.
Displaying images is a bit tricky in mkdocs wrt markdown.

closed 2025-01-30 by pattacini