mesh-iit/documentation #383: Document how the ansible installation process should be performed on the icub-head/ergocub-torso

traversaro, 2025-01-02:
See https://github.com/icub-tech-iit/documentation/pull/381#discussion_r1889851961 .

pattacini, 2025-01-02:
/remind January 10 @davidelasagna

traversaro, 2025-01-29:
In general this issue tracks the documentation of how to follow the documented steps when no mouse/keyboard/screen is connected to the board, but in particular I see two slightly different use cases it could make sense to document:
* A new robot gets manufactured, and it gets a brand new icub-head/ergocub-torso board. How is the Ubuntu installed in that case, when no mouse/keyboard/screen is connected to the board?
* An old robot has an older ubuntu distro (like Ubuntu 20.04) and you want to upgrade to a newer distro (for example as the latest robotology distro does not support the older ubuntu distro anymore). Given an existing robot, how can I upgrade Ubuntu distro, when no mouse/keyboard/screen is connected to the board?